udacity/fcnd-issue-reports #7: Lesson 2.8 Batteries - Provide resources

jtuckerdunn, 2018-01-24:
https://classroom.udacity.com/courses/ud787-preview/lessons/35850006-650e-483f-a46f-c3a58809cbf0/concepts/bdb44663-ed36-4d88-a505-4ad451e7cc88

The quiz makes this statement "Do a bit of research online to answer the following question:"

I think it would be of better value to the student if we provide recommended articles/resources for them to read such as in lesson 2.6 Motors instead of telling them to look it up themselves.

jtuckerdunn, 2018-02-09:
Students have also requested additional links/resources to propellers, IMUs, and GPS

delvingdeep, 2018-02-10:
I think one or two starter links should be provided, and rest should be left upon students to do some background research on their own. This way they can gain some additional information based upon their interest. 

Suggestion:
Can we keep a  prompt type of response collector, requesting 3 or 5 external links from student? I think that would serve the purpose of getting them do some research.

jtuckerdunn, 2018-04-23:
Will keep quiz as is to encourage student research.

closed 2018-04-23 by jtuckerdunn