udacity/fcnd-issue-reports #10: [Simulator] Arm / Disarm Button and Guided / Manual Button don't change state if controled by code

silentwater, 2018-01-25:
As in the Title already written. If controling the Simulator programmatically and arming / disarming / taking control by code, the buttons state don't change.

domluna, 2018-02-06:
I'm unable to reproduce this on the MacOS version of the latest release binary https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.0.3-preview

silentwater, 2018-02-06:
nickfixit in the Slack Channel has the same problem that the Disarm / Arm Buttons are not working if controlling the sim programmatically. Anyway, no big issue, just a minor bug at pressent as log as the sim react at all.

domluna, 2018-02-06:
There was a bug in the earlier versions but it should be fixed in the latest version so I'm not sure what's going on. Is there a deterministic way to break it you can share?

silentwater, 2018-02-07:
I just removed the whole conda fcnd environment and also deleted the starter kit and udacidrone.

I did the whole setup again from the beginning as described and also downloaded the new simulator version 0.0.3 but I still have the same problem. The Arm / Disarm and Manual / Guided Buttons are not changing state. Furthermore I noticed also the Latitude / Longitude / Altitude display is not working.

In the attached picture the quad is flying an the rotors are spinning, but one can see that the Buttons and the Latitude /... display is still showing 0.

Furthermore I attached the TLog.txt file if it is of any help. If I can provide any other log files or anything, just let me know.
[image]
[TLog.txt](https://waffleio-direct-uploads-production.s3.amazonaws.com/uploads/584f133f103fea0f000d78fa/125516c66e82c728ace21e0d46dcb89524c0cfeb9ffbb049a15ee4d6730472727f55f07321d7ba7ae64723505f5f4cab120b094cfffb9b37b4f46d2ede0356b49b381d66cc2f10abb2831abe09ba21834d86b0b88248d2a5.txt)

asimonov, 2018-02-07:
I have this problem on MacOS with version 0.0.3

asimonov, 2018-02-07:
same problems as silentwater describes

domluna, 2018-02-08:
@asimonov what macos version? I'm on the high sierra 10.13.3 and the binary works fine

asimonov, 2018-02-08:
I am on 10.11.6

asimonov, 2018-02-08:
in terms of config -- I have created fcnd env as described in lectures. think I have done this couple of weeks ago. using locally installed miniconda

silentwater, 2018-02-08:
Maybe it is possible to provider a Simulator / udacidrone with extended Logging Options, eventually with the possibility to set different log levels.

If there is any posibility to provide more detailed Log files, just let me know, so I can attach it here.

silentwater, 2018-02-08:
Just another thing which I noticed. If I press Shift-R within the Simulator to Reset the scene, the drone doesn't react at all anymore to any commands from the python / ipython console. Even take_off or cmd_position doesn't work anymore, not only the arm/disarm/... buttons not changing state.

I have to restart the Simulator to get it working again.

asimonov, 2018-02-10:
I have just downloaded version of Simulator for Path Planning and it works as expected. I see coordinates updated from the start. And when script sends Arm/Control the indicators in simulator change as expected. I did not do any changes in my laptop. Not even rebooted since when I reported issues with simulator.0.0.3 version.

domluna, 2018-02-10:
https://docs.unity3d.com/Manual/LogFiles.html . This link should be added somewhere in the classroom since it could be useful for debugging sim issues. Player.log is the one of interest.

silentwater, 2018-02-11:
I attached the "output_log.txt" file from my Windows 7 Machine which according to the unity3d manual should correspond to to Player.log.

There are some error messages, but can't tell if it has anything to do with my Arm / Manual Button not working and the Latitude / Longitude /... and the compas not working.

See attached Log File:
[output_log.txt](https://waffleio-direct-uploads-production.s3.amazonaws.com/uploads/584f133f103fea0f000d78fa/125516c66e82c728ace21e0d46e7818e3397979683e6b801e81fe086264c22703d46f9692584b579ec1a2a53565d47f949595f44fefeda78f0be3f72984019af80330b2b9b645bf9fa9015ad00bd2a8a4a84bebf8f4bc5eb3a25e78f9d02.txt)

The file
C:\buildslave\unity\build\artifacts/generated/common/runtime/DebugBindings.gen.cpp

mentioned in the loge file doesn't exist on my machine. Also the whole folter structure doesn't exit.

silentwater, 2018-02-11:
I think I found the problem.

First I was looking on Google for the error message "XInput1_3.dll not found. Trying XInput9_1_0.dll instead..." in output_log.txt, which hinted me to install the DirectX drivers. But doing so didn't fix the problem.

So I was looking into the error message 

ArgumentException: Toggle Best (UnityEngine.UI.Toggle) is not part of ToggleGroup Quality (UnityEngine.UI.ToggleGroup)
  at UnityEngine.UI.ToggleGroup.ValidateToggleIsInGroup (UnityEngine.UI.Toggle toggle) [0x00036] in <a7d1aa3efd4747538d44d2a67feaa4b9>:0 
  at UnityEngine.UI.ToggleGroup.NotifyToggleOn (UnityEngine.UI.Toggle toggle) [0x00001] in <a7d1aa3efd4747538d44d2a67feaa4b9>:0 
  at DroneUI.Awake () [0x0005b] in <554d77d7643940a29703d21bd6c6a509>:0

and was playing around with the quality settings in the simulator.

In the Screenshot you can see that the quality settings are in an illegal state. According to the screenshot Quality 'Simple' and 'Best' are selected at the same time.
[image]

So I changed the quality to 'Simple' but nothing changed except the grafics quality. But when I restarte the simulator, suddenly the Latitude / Longitude /... display and also the Buttons have been working again.

Than I changed the quality back to ''Best' and restarted and the buttons have been again in the illegal state and things again not woking. Than I tried 'Good' and restartet and things have been working again.

All the time if I change the Quality Settings to 'Best' and restart the simulator, the Quality Buttons end up in an illegal state having 'Simple' and 'Best' selected at the same time and the Coordinate Display and the Buttons not working by using a python script.

There seems to be a bug with the qualit sate settings. If I use 'Simple' or 'Good' everything is working fine independent of the resolution I choose.

domluna, 2018-02-12:
This should be fixed on the next release

domluna, 2018-02-15:
can you try on this release https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.0.4-preview?

asimonov, 2018-02-15:
works for me on MacOS 10.11.6!

silentwater, 2018-02-17:
Thx, working now also on my Win7 machine. Think can be closed ass fixed :)

jtuckerdunn, 2018-02-24:
Closing as fixed in release v0.0.4

closed 2018-02-24 by jtuckerdunn