udacity/fcnd-issue-reports #11: Controling the simulator by code within the ipython console requires the console to be starte with admin rights

silentwater, 2018-01-25:
As statet in the Title. If one wants to control the Drone Simulator by code within the ipython console prompt, the console needs to be started with administration rights. (Testet on Windows 7).

Either need to mentions this in the course in the part where the setup is explained or need to change the code, so no admin rights are needed.

domluna, 2018-01-25:
I'm not familiar with Windows 7, is it difficult to start with admin rights? I'm not sure if there's anything we can do avoid that step.

silentwater, 2018-01-25:
No, not difficult. Just need to right click on the Programm you want to start and in the up comming context menu select "Run as Administrator". Even one can't avoid this issue, it would be good to mention this in the course description for setting up the environment for Windows Users in case people are running into problems. If people run the console without admin rights and try to control the drone, they get strange error messages which is hard to figure out that it is because of missing admin rights.

silentwater, 2018-01-25:
The program is using port 5760 for the tcp connection. I just know that certain port ranges in Windows are restricted and can only be used with Administration privileges. Which port ranges are restricted depends also on the Windows Version. Maybe thats the issue why Admin rights are needed. Just guessing, not sure...

domluna, 2018-02-02:
@jtuckerdunn should the admin note be added in the classroom or on the github repo README?

jtuckerdunn, 2018-02-02:
@domluna Let's just add id to the README

domluna, 2018-02-05:
Added a note to the starter kit readme.

closed 2018-02-05 by domluna