udacity/fcnd-issue-reports #17: fcnd environment setup issue

delvingdeep, 2018-01-31:
Error reads: 

"Failed to build wheel for pymavlink.
Failed to build pymavlink." 

Link to slack post for screenshot of the error: https://udacity-flying-car.slack.com/files/U76UK7NNL/F91F0JQAZ/error_pymavlink.png

This is the error I see while creating conda environment on my macOS. I read post about same error encountered on windows too. 

Is there any package missing in the env list?

jtuckerdunn, 2018-01-31:
This error has been encountered by multiple people. I believe it is caused because the future module is built out of order. Pymavlink successfully builds later in the setup process but we should still get rid of this error so it doesn't confuse students during setup.

delvingdeep, 2018-02-01:
Thanks @jtuckerdunn for clarification!!

domluna, 2018-02-02:
I'm not sure we can avoid the error completely without having two separate scripts. By that I mean 1 where `future` and installed and then call another script to install `pymavlink`. AFAIK `pymavlink` still successfully builds despite the error message.

jtuckerdunn, 2018-02-02:
I think it would be worth it to find a way to get rid of the error message. Even if we note that an error occurs I think it will cause confusion.

delvingdeep, 2018-02-03:
I think till the time we find a solution to it, we shall add a note about this error either in the classroom or in fcnd environment github repo to avoid creating any confusion.

domluna, 2018-02-05:
Added a note to the starter kit repo, I think we should keep this open for now.

delvingdeep, 2018-02-05:
Thanks @jtuckerdunn and @domluna for updating docs about this error.

4FH, 2018-02-06:
python -c 'import pymavlink'    --  needs double quotes. Single quotes version returns "SyntaxError : EOL while scanning string literal" (Win10, Anaconda prompt)

4FH, 2018-02-06:
also, numpy above 1.12.0 wont work for unknown reason on Win10.  fcnd yml installs 1.14.0. If you try to import numpy or even execute first line "from udacidrone..." it will throw bunch of errors. you need to manually uninstall numpy and install 1.12.0.

>>inside activated fcnd environment:
pip uninstall numpy
pip install -Iv numpy==1.12.0

I haven't tried to do uninstall /install through conda though.  Anybody?

4FH, 2018-02-06:
Conda version of above works too:
//conda uninstall numpy  -- not sure if it is needed
conda install numpy=1.12.0
(it will warn about downgrading from 1.14 and will install intel-openmp and mkl packages)

domluna, 2018-02-06:
@4FH thanks for reporting this

4FH, 2018-02-06:
Also, I saw environment creation failing on Win7 because miniconda didn't have git installed. Cleaned up and deleted fcnd env, then "conda install git" and rerun conda env create successfully

4FH, 2018-02-06:
weirdly enough, I saw same issue with numpy 1.14 import on Win7 box too. After some digging issue seems to be with absence of mkl packages that numpy requires and wrong(?) numpy build for current conda release. after I did :
conda install mkl
conda install numpy

everything is ok. on last one there was downgrade warning:

>>The following packages will be DOWNGRADED:

    numpy:  1.14.0-py36_blas_openblas_200 conda-forge [blas_openblas] --> 1.14.0-py36h4a99626_1

Its quite puzzling

davinderc, 2018-02-06:
I don't think pymavlink installs properly on Ubuntu, but I'm checking that more thoroughly now.

davinderc, 2018-02-06:
So from what I can see, future does not get installed before pymavlink on Ubuntu. I tried looking through the environment.yml file to see if I can install pymavlink after the env is created, but it seems that's done somewhere else.

domluna, 2018-02-06:
@davinderc what's the output of

```
python -c "import pymavlink"
```

davinderc, 2018-02-06:
@domluna `ModuleNotFoundError: No module named 'pymavlink'`

davinderc, 2018-02-07:
@domluna it's my impression that pymavlink 2.2.0 does not install happily on Ubuntu 16.04. Is it possible to use 2.2.8? I was able to install that version easily, but then when I try to install udacidrone, it tries to roll it back to 2.2.0, even with future and lxml installed. 

I also tried to install 2.2.0 manually with pip with the following results:

```pip install pymavlink==2.2.0
Collecting pymavlink==2.2.0
  Using cached pymavlink-2.2.0.tar.gz
Requirement already satisfied: future in /home/controlfreak/miniconda3/envs/fcnd/lib/python3.6/site-packages (from pymavlink==2.2.0)
Requirement already satisfied: lxml in /home/controlfreak/miniconda3/envs/fcnd/lib/python3.6/site-packages (from pymavlink==2.2.0)
Building wheels for collected packages: pymavlink
  Running setup.py bdist_wheel for pymavlink ... error
  Complete output from command /home/controlfreak/miniconda3/envs/fcnd/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-5o17zf8j/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp1y2mteg9pip-wheel- --python-tag cp36:
  running bdist_wheel
  running build
  running build_py
  No XML message definitions found
  
  ----------------------------------------
  Failed building wheel for pymavlink
  Running setup.py clean for pymavlink
Failed to build pymavlink
Installing collected packages: pymavlink
  Found existing installation: pymavlink 2.2.8
    Uninstalling pymavlink-2.2.8:
      Successfully uninstalled pymavlink-2.2.8
  Running setup.py install for pymavlink ... error
    Complete output from command /home/controlfreak/miniconda3/envs/fcnd/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-5o17zf8j/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-p3fvgxc_-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    No XML message definitions found
    
    ----------------------------------------
  Rolling back uninstall of pymavlink
Command "/home/controlfreak/miniconda3/envs/fcnd/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-5o17zf8j/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-p3fvgxc_-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-5o17zf8j/pymavlink/
```

mkrauter, 2018-02-07:
On Ubuntu 16.04 Conda environment creation fails at installing 'pymavlink' with 'ModuleNotFoundError: No module named 'future'' and 'No XML message definitions found'. This error is not recovered later, i.e. 'import udacidrone' would lead to 'No module named 'udacidrone''.
Quick'n'dirty workaround:
Create your environment: 'conda env create -f environment.yml', it will fail.
Clone udacidrone: 'git clone https://github.com/udacity/udacidrone.git'
Edit 'udacidrone/setup.py', modify 'pymavlink==2.2' to 'pymavlink==2.2.8' at section 'install_requires'
Source environment: 'source activate fcnd'
Install udacidrone from local source: 'pip install udacidrone/'

davinderc, 2018-02-07:
That fixed it! Thanks @mkrauter I didn't realize I could do it from the setup.py file

domluna, 2018-02-07:
I'll bump it up to 2.2.8

4FH, 2018-02-08:
Is there any particular reason to install packages from "conda-forge" channel?
After digging through dependency puzzle on Windows, here is preliminary outcome:
1. conda-forge version of numpy is compiled for openblas support. Requires openblas.dll inside conda-forge openblas package.
2. open_blas package needs fortran dlls: flang.dll and flangrti.dll (conda-forge libflang package)
3. Both fortran dlls  need **libomp.dll** from **conda-forge openmp** which **never got installed** because libflang json file from conda-forge do not list dependency on it. Instead it installs ompstub.dll that never get used. Either json need to be corrected to include extra dependency for conda-forge openmp package, or both fortran dlls where compiled with wrong settings (wild guess they should've use ompstub, as to my knowledge you need openmp when using "normal" numpy compiled for intel mkl libs.)

__TLDR;__  Short term solution: __conda install -c conda-forge openmp__

Otherwise conda-forge numpy won't import under windows(including all other packages that are depending on it).  Version downgrading that I tried before (see above posts), worked only because it was pulling numpy from default (non conda-forge) channel and dependencies where referenced and installed correctly (mkl and openmp)

delvingdeep, 2018-02-12:
With the modification to `environment.yml` file, now there are no error message while setting up FCND conda environment. 
It works well on macOS.

closed 2018-02-13 by domluna