udacity/fcnd-issue-reports #18: 2.10 QUIZ wrong answer

dambatenne, 2018-01-31:
Increase thrust on A and D while decreasing thrust by the same amount on B and C.
A and D rotate clockwise, so angular force related to them is counterclockwise. So the answer should be: Quad yaws counterclockwise as seen from above.

4FH, 2018-02-01:
Correct. Engine/propeller torque results in spin _opposite_ to rotational direction of propeller, at a lower rotational speed due to higher moment of inertia (assuming that the body of quad is heavier than propeller). Funny enough,  misconception in question somehow spread all over internet and found its way even to some university courses.

ryan-keenan, 2018-02-02:
Great catch!  Thanks for reporting. Fixed now :)

closed 2018-02-02 by ryan-keenan