udacity/fcnd-issue-reports #20: 2.7 Propeller Quiz, need more preconditions

4FH, 2018-02-01:
Test do not clarify conditions of flight, efficiency definition etc. Whole thing is highly debatable. For example, in hovering or at very low speeds, high pitch props are very inefficient due to stalling. Large diameter props statement deserve to be all around "best efficiency" choice, if all other parameters are equal. Low pitch props provide best efficiency (and thrust) while hovering. Not so much at a greater speed. And so on.

jtuckerdunn, 2018-02-09:
Classroom link: https://classroom.udacity.com/nanodegrees/nd787-beta/parts/38fbf270-3dd1-43e8-bf32-fc6bef248454/modules/129fb574-4a4a-4d07-a084-bc2855610afe/lessons/99905423-0638-479b-a9f5-d5f33de6c613/concepts/5e3a8b4c-c1e7-4cb3-95fa-9f2540f5d50c

sabondano, 2018-02-16:
What's the story here? What's the right set of answers?

ryan-keenan, 2018-03-16:
I got rid of this quiz for now as it seems it is not suitable for a simple matching game.  We can revisit if it seems like there's something better to include here.

closed 2018-03-16 by ryan-keenan