udacity/fcnd-issue-reports #25: 32-bit Linux executable

davinderc, 2018-02-06:
I'm not sure if this is a bug, but I've been trying to run the Linux executable on Ubuntu 16.04 with no success. After realizing it was compiled for a 32-bit system I installed libncurses, libc6 and libstdc++6 for 32-bit archs, and now I get "SIGABRT" before it terminates.

Maybe I'm missing something, but I'm sure this executable should be compiled to 64-bit archs these days.

domluna, 2018-02-06:
@davinderc I'll compile 1 for 64-bit Linux, I thought that was the default but apparently, it's not.

domluna, 2018-02-06:
Updated with a universal zip, this 1 should work https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.0.3-preview

davinderc, 2018-02-06:
Yep! It runs now, thanks! I'm not sure why the 32-bit one wouldn't run with the required libraries, since I can run the car simulators just fine (although maybe those are all in 64 bit already).

closed 2018-02-06 by domluna