udacity/fcnd-issue-reports #30: error while setting up fcnd environment

AndyAtUdacity, 2018-02-07:


AndyAtUdacity, 2018-02-07:
This was reported by a student ( #diyjac in Slack)

domluna, 2018-02-07:
if this is on ubuntu try `sudo apt-get install libxml2-dev libxslt-dev`

davinderc, 2018-02-07:
This method seems to resolve the missing `future` module, but the pymavlink 2.2.0 seems to still have an issue installing: https://github.com/ArduPilot/pymavlink

domluna, 2018-02-07:
Does changing pymavlink to 2.2.8 fix this?

davinderc, 2018-02-07:
I've got it up and running now it seems (pymavlink and udacidrone both import without a word of warning). Haven't checked with the simulator though.

domluna, 2018-02-07:
tested this on 16.04 with pymavlink 2.2.8 and it works so I'm closing this. The simulator issues https://github.com/udacity/fcnd-issue-reports/issues/26, etc are being worked on separately.

closed 2018-02-07 by domluna