udacity/fcnd-issue-reports #40: docs - visualization

asimonov, 2018-02-08:


asimonov, 2018-02-08:
here is the description:

in docs here: https://udacity.github.io/udacidrone/docs/visdom-tutorial.html

'While the FCND simulator provides plotting functionality out of the box' -->>> I currently cannot see any command in simulator to do plotting.

'Prior to plotting anything we have to start the visdom server'  ->>>> visdom is not part of fcnd environment at the moment

asimonov, 2018-02-08:
I also needed to tweak the code around numpy arrays. here is my working solution snippet:

        # visualisation using visdom
        self.init_visdom()
        self.register_callback(MsgID.LOCAL_POSITION, self.update_pos_plots)
...

    def init_visdom(self):
        self.v = visdom.Visdom()
        assert self.v.check_connection()
        p = self.target_position # use target here as home is not set yet
        # Plot NE
        ne = np.array([p[0], p[1]]).reshape(-1, 2)
        self.ne_plot = self.v.scatter(ne, opts=dict(
            title="Local position (north, east)",
            xlabel='North',
            ylabel='East'
        ))
        # Plot D
        d = np.array([-1*p[2]])
        self.t = 1
        self.d_plot = self.v.line(d, X=np.array([self.t]), opts=dict(
            title="Altitude (meters)",
            xlabel='Timestep',
            ylabel='Down'
        ))

    def update_pos_plots(self):
        self.t += 1
        p = self.local_position
        self.v.scatter(np.array([p[0], p[1]]).reshape(-1, 2), win=self.ne_plot, update='append')
        self.v.line(np.array([-1*p[2]]), X=np.array([self.t]), win=self.d_plot, update='append')

jtuckerdunn, 2018-02-09:
The in-simulator plotting is still a work in progress.

closed 2018-02-09 by jtuckerdunn