udacity/fcnd-issue-reports #45: Simulator fails to respond after reset

davinderc, 2018-02-09:
While working on the project, sometimes I had code that made the drone go crazy, so I would exit with CTRL+C and reset the simulator with SHIFT+R as mentioned in the controls menu. However, running the code again after this resulted in nothing happening with the simulator. The drone just sat there, didn't arm or anything.

berndrietberg, 2018-02-11:
I have the same thing. Have to restart the drone-simulator program everytime I want to retest my code.
Maybe a Drone.reset function?

domluna, 2018-02-11:
We have a fix for this it’s just not a build yet

domluna, 2018-02-15:
This should work on this release https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.0.4-preview

jtuckerdunn, 2018-02-24:
Closing this issue assuming it was fixed in release v0.0.4

closed 2018-02-24 by jtuckerdunn

mimoralea, 2018-03-01:
Maybe something similar. In my Linux machine I can't control the drone manually, on my Windows is all fine.

I arm the drone, press the spacebar and it goes crazy. ~5 seconds later is -1000 meters on altitude.

[image]