udacity/fcnd-issue-reports #50: Cannot connect to the simulator

ubaierbhat, 2018-02-10:
I cannot connect to the FCND simulator. I see the following error when I run `drone.start()` 

Traceback (most recent call last):
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/site-packages/udacidrone/connection/mavlink_connection.py", line 107, in dispatch_loop
    msg = self.wait_for_message()
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/site-packages/udacidrone/connection/mavlink_connection.py", line 193, in wait_for_message
    msg = self._master.recv_match(blocking=True, timeout=1)
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/site-packages/pymavlink/mavutil.py", line 347, in recv_match
    m = self.recv_msg()
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/site-packages/pymavlink/mavutil.py", line 310, in recv_msg
    s = self.recv(n)
  File "/Users/ubaierbhat/miniconda3/envs/fcnd/lib/python3.6/site-packages/pymavlink/mavutil.py", line 968, in recv
    data = self.port.recv(n)
ConnectionResetError: [Errno 54] Connection reset by peer

domluna, 2018-02-10:
@ubaierbhat 

1. Is the simulator running?
2. What host, port are you using for `MavlinkConnection`?

ubaierbhat, 2018-02-10:
I just managed to connect now after restart everything couple of times.

The simulator was running and host port was 127.0.0.1:5760

closed 2018-02-13 by domluna