udacity/fcnd-issue-reports #51: 2.1.10 Breadth vs. Depth solution

davinderc, 2018-02-11:
The Breadth-first solution is given (the first 5 items in the exercise are for this), but nothing is mentioned about depth-first afterwards. It would be good to offer at least a hint on the depth-first solution.

davinderc, 2018-02-11:
Actually, nevermind, my Jupyter notebook was behaving strangely but after a restart it returned the path. From my understanding, the difference between the two is simply popping the first vs the last elements in the queue.

domluna, 2018-02-15:
> From my understanding, the difference between the two is simply popping the first vs the last elements in the queue.

Right. DFS explores each path in its entirety before moving on whereas BFS explores all closest neighbors first, then expands to search neighbors of neighbors, etc.

The notebook has been updated to reflect the focus on BFS.

closed 2018-02-15 by domluna