udacity/fcnd-issue-reports #56: Quaternion exercise is not clear

orenmeiri, 2018-02-15:
In the current notebook (Vehicle Representation -> 14. Quaternion notebook) it is not clear what is expected from student. It is not a solution nor problem to solve, seems half baked at this stage.

asimonov, 2018-02-18:
yes, I agree. Just gone through the video 10: quaternions. understood that there is much more there than meets the eye. so before jumping to quaternions notebook i spent a few hours going through two articles on wikipedia:

https://en.wikipedia.org/wiki/Quaternion

https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation

I suggest at least the links to them are incorporated into the lecture notes. or better some of the wiki intuitions put into video.

PS: my background is applied math undergrad degree 17 years ago. but wiki articles made a great job linking everything together. so I would say the current videos are a very good motivation for the exploration to start but are not sufficient.

jtuckerdunn, 2018-02-20:
It might make sense to add written content going more in-depth describing quaternions in between the video and excercise. The wikipedia resources linked above are great.

ryan-keenan, 2018-02-25:
Thanks for the suggestions guys!  This exercise has been updated with some external references (including a music video) and I think is better now.  Keep the feedback coming! :)

closed 2018-02-25 by ryan-keenan