udacity/fcnd-issue-reports #61: Symantec AV blocks car simulator

TanayAarush, 2018-02-16:
AVs are necessity of the day. But I am unable to run the flight simulator (fcnd-sim_windows.exe) in my Windows7 PC. Symantec is deleting the file as it is 'untrustworthy'. Because they know the file for only 1 Month and Only 50 Symantec antivirus users have encountered the file. Exception list do not work. Its too risky to disable or uninstall AV.

Is it possible for Udacity to inform Symantec to trust the file.

jtuckerdunn, 2018-02-20:
This issue has also occurred with other antivirus software.

jtuckerdunn, 2018-02-24:
Students are being directed to report these antivirus blocks as false positives.

demi180, 2018-03-03:
Unfortunately we can't directly tell or ask them to trust us, but we will look into ways to have the app code-signed. In the meantime please continue reporting as false positive.

closed 2018-04-21 by domluna