udacity/fcnd-issue-reports #63: bresenham explanation is wrong

asimonov, 2018-02-18:
in bresenham video the algorithm seem to be explained incorrectly. if implemented as stated it will always draw either diagonal line or horizontal line. if you look at here: http://www.roguebasin.com/index.php?title=Bresenham%27s_Line_Algorithm#Python you see that the right algorithm tracks error term between steps which determines when Y should step up

jtuckerdunn, 2018-02-20:
The referenced video is 3.8 video: bresenham

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/13d546b4-4f46-4d24-8297-799fe73fd987

ryan-keenan, 2018-02-20:
Thanks for reporting this @asimonov !  I agree this is a poor explanation but unfortunately we don't have Nick around right now to redo it.  Instead, we will be adding a better explanation to the text that follows. I'll update this ticket when we do!

ryan-keenan, 2018-03-15:
Classroom explanation is updated and seems to work well.  I'm working on a rewrite of Nick's script to patch the issues in the video.

ryan-keenan, 2018-03-15:
Another student pointed out we could be doing better and eliminating floating point values (the slope, m) entirely from the algorithm.  The classroom solution has been updated to reflect this.

ywiyogo, 2018-04-15:
@ryan-keenan did you update the classroom explanation in the lesson 7.8 (exercise explanation) or lesson 7.7 under the video? I cannot find any explaination under the video in 7.7.

ryan-keenan, 2018-04-16:
The text explanation in the exercise page has been updated (and the exercise itself). The video will be updated soon, thanks!

closed 2018-04-16 by ryan-keenan