udacity/fcnd-issue-reports #64: links to solution notebooks dont work

asimonov, 2018-02-18:
for example if I click link inside bresenham notebook to 'solution' it shows me an error message. not sure if it should work in beta or only supposed to work in final FCND

delvingdeep, 2018-02-19:
[image]

Shows the above error message

ryan-keenan, 2018-02-24:
Thanks Deep!  Should be fixed now for all notebooks

closed 2018-02-24 by ryan-keenan