udacity/fcnd-issue-reports #67: Broken Link: Collinearity_Condition-starter Solution Link

matthewzimmer, 2018-02-19:
Classroom:

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/286f789f-bf70-4896-b41c-68907a0c9825

Appears to be broken:

https://cocoview9d7db0d8.udacity-student-workspaces.com/notebooks/Collinearity-Solution.ipynb

matthewzimmer, 2018-02-19:
[image]

jtuckerdunn, 2018-02-20:
This issue is being tracked in #64

closed 2018-02-20 by jtuckerdunn