udacity/fcnd-issue-reports #72: Audio pitch on video is low

JonathanKSullivan, 2018-02-20:
Audio pitch on 3D Motion Planning > Graph Representation > (Videos: 2,3)

jtuckerdunn, 2018-02-20:
This references the waypoint extraction and collinearity videos. The audio has very low pitch.

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/417c5bf3-e35b-49bb-8967-5b923d778813

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/85f5ff5b-5ab1-4daf-bb43-eb1d55179449

jtuckerdunn, 2018-02-20:
pitch is also high in this video 2.3.15 grids to graphs

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/dfeadb89-031e-4166-a9bf-ce78415d7cb7

ryan-keenan, 2018-02-20:
Thanks for reporting this @JonathanKSullivan !  In the next couple days we will be replacing these videos with final cuts where any audiovisual glitches should be ironed out.  I'll update this ticket when we do!

ryan-keenan, 2018-02-24:
Fixed now!

closed 2018-02-24 by ryan-keenan