udacity/fcnd-issue-reports #73: Geodetic Frame video is unavailable

jtuckerdunn, 2018-02-20:
The geodetic frame video 2.2.2 shows the error "this video is unavailable" 

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/91ccafae-d090-4144-a8f3-06cb8ac10757/concepts/bb5c6d71-c31b-4b99-944d-0d315769f9ed

jtuckerdunn, 2018-02-20:
2.2.12 motions as transformations has same error

https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/91ccafae-d090-4144-a8f3-06cb8ac10757/concepts/ce9bcf88-f75e-4cf3-8a7d-d20ab2bf104c

ryan-keenan, 2018-02-20:
Thanks for catching this! Fixed now.

closed 2018-02-20 by ryan-keenan