udacity/fcnd-issue-reports #79: Below the Bresenham video we can have explanation.

Sagar-Jarvis, 2018-02-21:
We can have an explanation of the algorithm below the video itself. Video just does the overview but with explanation below the video will help. An explanation something like this: 

https://www.cs.helsinki.fi/group/goa/mallinnus/lines/bresenh.html

Sagar-Jarvis, 2018-02-21:
Or something like this: https://www.tutorialspoint.com/computer_graphics/line_generation_algorithm.htm

jtuckerdunn, 2018-02-24:
https://classroom.udacity.com/nanodegrees/nd787-beta/parts/1822bb5b-8b4b-496e-80b5-ab74d031113b/modules/8c3efada-ac9a-4a6a-bc77-4036137f56c6/lessons/5a623140-3e28-4ef1-a0f2-104715381c54/concepts/13d546b4-4f46-4d24-8297-799fe73fd987 Link to video. It looks like the helsinki link is already referenced in the Bresenhem exercise.

ryan-keenan, 2018-03-01:
yes, now the helsinki link is in there and there's a better discussion of what we're implementing as well as the Python package version of Bresenham. Thanks for suggestions!

closed 2018-03-01 by ryan-keenan