udacity/fcnd-issue-reports #87: 2.1.6 Partial plans quiz answer confusing

jtuckerdunn, 2018-02-22:
The quiz answer for partial plans is confusing and not super clear.
"
That's correct! From the start state, you can move in 3 different directions, giving you 3 partial plans, but moving across the grid, you'll only be expanding 2 out of those 3 plans, rather than adding new ones. On the other side, you'll get one more plan by expanding into the cell between the obstacle and the goal, for a total of 4."

ryan-keenan, 2018-02-25:
Thanks for the feedback, I've tried to clean this up and I think the wording is better now, though it's hard to explain in words.  I'd like to ultimately add a gif there that just shows what's happening. Not hard to do, just will take a little time.  I'll close this for now and we can revisit if it persists as a problem.

closed 2018-02-25 by ryan-keenan

jwdunn1, 2018-03-01:
The updated text is not quite right: "But next can only expanding 2 out of those 3 plans because of the obstacle." Perhaps you meant to drop the "ing". 

Also, I thought the answer was 12, so it's still not clear (to me).