udacity/fcnd-issue-reports #96: FCND-Sim_MacOS North, East values sometime takes a wild jump run with BackyardFlyer

Mark70117, 2018-02-25:
**What is going wrong**

North, East values sometime takes a wild jump FCND-Sim_MacOS w/ BackyardFlyer

Logs/TLog.txt

MsgID.LOCAL_POSITION,1519579443.7454259,0.0117450,0.0343865,0.0060761
MsgID.LOCAL_POSITION,1519579443.7474129,0.0320142,-0.1255525,-0.0253832
MsgID.LOCAL_POSITION,1519579443.7493670,0.1065134,-0.0743402,-0.0493281
MsgID.LOCAL_POSITION,1519579443.7825401,4157078.2500000,-10843894.0000000,-0.0729923
MsgID.LOCAL_POSITION,1519579444.0318999,4157078.2500000,-10843894.0000000,-0.3936308
MsgID.LOCAL_POSITION,1519579444.2822411,4157078.0000000,-10843895.0000000,-1.1239108

**How to recreate it**

  My environment
    macOS High Sierra version 10.13.3
    Mac mini (Mid 2011) 8GB
    FCND-Sim_MacOS.app v 0.0.4 https://github.com/udacity/FCND-Simulator-Releases/releases
    https://github.com/udacity/udacidrone 3bfcb1d
    https://github.com/udacity/FCND-Backyard-Flyer solution branch  2a71eb8

I wrote some scripts to repeatedly run FCND-Sim_MacOS and FCND-Backyard-Flyer.
Left to run overnight 4-20% of the runs fail.  I was able to increase the percentage
by opening many more application on the machine running FCND-Sim_MacOS

timeOut.sh -- script to timeout another script, added screencapture on failure
singleRun.sh -- script to run BackyardFlyer once with freshly opened FCND-Sim_MacOS.app
thousandRun.sh -- script to run singleRun.sh 1,000 times
untilCrashRun.sh -- script to run singleRun.sh until new failure screencapture

---
$CMD cat timeOut.sh
```#!/bin/bash
## Based on https://stackoverflow.com/questions/687948/timeout-a-command-in-bash-without-unnecessary-delay
## add screencapture $$.jpg
#
# The Bash shell script executes a command with a time-out.
# Upon time-out expiration SIGTERM (15) is sent to the process. If the signal
# is blocked, then the subsequent SIGKILL (9) terminates it.
#
# Based on the Bash documentation example.

# Hello Chet,
# please find attached a "little easier"  :-)  to comprehend
# time-out example.  If you find it suitable, feel free to include
# anywhere: the very same logic as in the original examples/scripts, a
# little more transparent implementation to my taste.
#
# Dmitry V Golovashkin <Dmitry.Golovashkin@sas.com>

scriptName="${0##*/}"

declare -i DEFAULT_TIMEOUT=9
declare -i DEFAULT_INTERVAL=1
declare -i DEFAULT_DELAY=1

# Timeout.
declare -i timeout=DEFAULT_TIMEOUT
# Interval between checks if the process is still alive.
declare -i interval=DEFAULT_INTERVAL
# Delay between posting the SIGTERM signal and destroying the process by SIGKILL.
declare -i delay=DEFAULT_DELAY

function printUsage() {
    cat <<EOF

Synopsis
    $scriptName [-t timeout] [-i interval] [-d delay] command
    Execute a command with a time-out.
    Upon time-out expiration SIGTERM (15) is sent to the process. If SIGTERM
    signal is blocked, then the subsequent SIGKILL (9) terminates it.

    -t timeout
        Number of seconds to wait for command completion.
        Default value: $DEFAULT_TIMEOUT seconds.

    -i interval
        Interval between checks if the process is still alive.
        Positive integer, default value: $DEFAULT_INTERVAL seconds.

    -d delay
        Delay between posting the SIGTERM signal and destroying the
        process by SIGKILL. Default value: $DEFAULT_DELAY seconds.

As of today, Bash does not support floating point arithmetic (sleep does),
therefore all delay/time values must be integers.
EOF
}

# Options.
while getopts ":t:i:d:" option; do
    case "$option" in
        t) timeout=$OPTARG ;;
        i) interval=$OPTARG ;;
        d) delay=$OPTARG ;;
        *) printUsage; exit 1 ;;
    esac
done
shift $((OPTIND - 1))

# $# should be at least 1 (the command to execute), however it may be strictly
# greater than 1 if the command itself has options.
if (($# == 0 || interval <= 0)); then
    printUsage
    exit 1
fi

# kill -0 pid   Exit code indicates if a signal may be sent to $pid process.
(
    ((t = timeout))

    while ((t > 0)); do
        sleep $interval
        kill -0 $$ || exit 0
        ((t -= interval))
    done

    # Be nice, post SIGTERM first.
    # The 'exit 0' below will be executed if any preceeding command fails.
    screencapture $$.jpg
    kill -s SIGTERM $$ && kill -0 $$ || exit 0
    sleep $delay
    kill -s SIGKILL $$
) 2> /dev/null &

exec "$@"
```

---
$CMD cat singleRun.sh
```
FCND_BACKYARD_FLYER_PATH=/Users/mranderso/UdacityFlyingCarNDClassroom/FCND-Backyard-Flyer/backyard_flyer.py
TIMEOUT_SCRIPT_PATH=~/tmp/Issue/timeOut.sh
APP_PATH=/Applications
APP_NAME=FCND-Sim_MacOS

killall ${APP_NAME}
sleep 2
open ${APP_PATH}/${APP_NAME}.app
sleep 20
{ time bash -x ${TIMEOUT_SCRIPT_PATH} -t 60 python ${FCND_BACKYARD_FLYER_PATH}  ;} > $$.time 2>&1
sleep 6
killall ${APP_NAME}
```

---
$CMD cat thousandRun.sh
```#!/bin/bash
SINGLERUN_SCRIPT_PATH=~/tmp/Issue/singleRun.sh

for i in {1..1000}
do
  sh -x ${SINGLERUN_SCRIPT_PATH}
done
```

---
$CMD cat untilCrashRun.sh
```#!/bin/bash
SINGLERUN_SCRIPT_PATH=~/tmp/Issue/singleRun.sh

ORIG_JPG_COUNT=`find . -maxdepth 1 -type f -name '*.jpg' | wc -l`
CURRENT_JPG_COUNT=`find . -maxdepth 1 -type f -name '*.jpg' | wc -l`

echo ${ORIG_JPG_COUNT}

while  [ $ORIG_JPG_COUNT -eq $CURRENT_JPG_COUNT ]
do
  sh -x ${SINGLERUN_SCRIPT_PATH}
  CURRENT_JPG_COUNT=`find . -maxdepth 1 -type f -name '*.jpg' | wc -l`
done
```

**My isolation efforts**

I added some print statement in MavlinkConnection.wait_for_message
(udacidrone/connection/mavlink_connection.py) and see that message
as parsed has the jump across space to (4157078.2500000,-10843894.0000000).

I also verified with wireshark that _msgbuf content matchs tcp traffic on port 5760, so
it appears that FCND-Sim_MacOS is sending the values and that they are not
being miscalculated on the receiving side.

`
90 {'_header': <pymavlink.dialects.v20.ardupilotmega.MAVLink_header object at 0x106937588>, '_payload': array('B', [255, 32,      0, 0, 18, 79, 0, 0, 88, 186, 125, 74, 247, 118, 37, 203, 198, 245, 168, 189, 136, 150, 145, 182, 0, 192, 120, 180, 96, 91, 2     9, 53]), '_msgbuf': array('B', [253, 28, 0, 0, 199, 0, 255, 32, 0, 0, 18, 79, 0, 0, 88, 186, 125, 74, 247, 118, 37, 203, 198,      245, 168, 189, 136, 150, 145, 182, 0, 192, 120, 180, 96, 91, 29, 53, 164, 88]), '_crc': 22692, '_fieldnames': ['time_boot_ms     ', 'x', 'y', 'z', 'vx', 'vy', 'vz'], '_type': 'LOCAL_POSITION_NED', '_signed': False, '_link_id': None, 'time_boot_ms': 20242     , 'x': 4157078.0, 'y': -10843895.0, 'z': -0.08250002562999725, 'vx': -4.338860890129581e-06, 'vy': -2.3166649043560028e-07, '     vz': 5.862002581125125e-07, '_posted': True, '_timestamp': 1519535288.114578}

91 fd1c0000c700ff200000124f000058ba7d4af77625cbc6f5a8bd889691b600c078b4605b1d35a458
`

Mark70117, 2018-02-27:
Reading API documentation today, I realize local NED origin is changed with set_home_position.  When investigating issue, I do recall sometimes seeing self.global_home values of [0 0 0] not in neighborhood of self.global_position (global_position [-121.9939587   37.4137477    0.54     ]).  Just adding that in, in case some situation like Drone instance not having yet received MsgID.GLOBAL_HOME before calling set_home_position would cause havoc.

Mark70117, 2018-02-27:
Modified the kill script to save Log/TLog.txt on a kill.   Ran again overnight last night.  49 failures, each with MsgID.STATE in TLog.txt before MsgID.GLOBAL_POSITION.

Change backyard_flyer.py 
```
        #self.set_home_position(self.global_position[0], self.global_position[1],
        #                       self.global_position[2])  # set the current location to be the home position
        self.set_home_position(0.0, 0.0, 0.0)
```
And see the same wild jump in local position to 4157078.2500000,-10843895.0000000
```
MsgID.LOCAL_POSITION,1519738817.6579180,0.1333989,-0.0053603,-0.0957332
MsgID.LOCAL_POSITION,1519738817.8602219,4157078.2500000,-10843895.0000000,-0.0900631
```

Mark70117, 2018-02-27:
checking for non zero values before using self.global_position in backyard_flyer.py has allowed the code to run 100 times without the wild jump in LOCAL_POSITION
```
    def arming_transition(self):
        print("arming transition")
        # begin ADDED code
        if self.global_position[0] == 0.0 and self.global_position[1] == 0.0: 
            print("no global position data, wait")
            return
        #end ADDED code
        self.take_control()
        self.arm()
        self.set_home_position(self.global_position[0], self.global_position[1],
                               self.global_position[2])  # set the current location to be the home position

        self.flight_state = States.ARMING
```

jwdunn1, 2018-03-01:
Mark70117 , excellent detective work! I was experiencing the same problem on Windows. Your sol'n for checking for non zero values during arming fixed the problem for me. Thank you!

domluna, 2018-03-11:
The home position is initially set in the sim to the initial drone location at the start of the scene. There was a bug where the longitude and latitude where passed in reverse, hence the crazy local position values. This is fixed in the newer sim releases. To get around this you would set the home position via udacidrone and wait for the change to propagate. In any case, this should no longer be an issue.

closed 2018-03-11 by domluna