udacity/fcnd-issue-reports #100: Udacidrone API Overview Reference Frame "Down (positive up)"

Mark70117, 2018-02-26:
https://udacity.github.io/udacidrone/docs/welcome.html

```Two different reference frames are defined and used within the Drone API. Global positions are defined as [longitude, latitude, altitude (positive up)]. Local reference frames are defined [North, East, Down (positive up)] and is relative to a nearby global home provided. ```

Down (positive up) seems confusing

jtuckerdunn, 2018-02-28:
Can we just add a sentence talking about why the reference frames are different? I think this is addressed in the classroom but it would be helpful here also.

Mark70117, 2018-02-28:
Just to be clear. The confusion I have is why not
Down (negative up)
or
Down (positive down)
why called Down if it is "(positive up)"

domluna, 2018-03-11:
cc @adrnp

adrnp, 2018-03-13:
unfortunate typo that is being fixed.  Should read `Down (positive down)`

jtuckerdunn, 2018-04-16:
@adrnp Has this been resolved?