udacity/fcnd-issue-reports #103: 3d project: shift+s in simulator is not producing new colliders.csv

asimonov, 2018-02-27:


demi180, 2018-03-03:
@asimonov Do you get the popup asking to save, as in the image I added? Or is that not coming up at all?
Could you please provide us the log file for the sim when you try this? They're located here:

macOS: ~/Library/Logs/Unity/Player.log
Windows: C:\Users\username\AppData\LocalLow\CompanyName\ProductName\output_log.txt
Linux: ~/.config/unity3d/CompanyName/ProductName/Player.log

[image]

asimonov, 2018-03-03:
No popup at all

_____________________________
Kind Regards,
Alexey Simonov

asimonov, 2018-03-03:
Will send logs later

_____________________________
Kind Regards,
Alexey Simonov

asimonov, 2018-03-03:
Right...
I re-downloaded the file FCND-Sim-macOS.app from beta project link.
It shows timestamp of 30dec17.

When I start backyard flyer i see the screen with the drone ok.
This is where I tried shift+s earlier.
Now I realise it was mistaken :)

When I start 3D motion planning project, however, the screen with the drone starts to flicker really quickly. And no controls work. All I can do it to kill Unity window.
Attaching the log for this case.
[Player.log](https://github.com/udacity/fcnd-issue-reports/files/1777874/Player.log)

asimonov, 2018-03-03:
I am on ElCapitan 10.11.6

domluna, 2018-03-11:
@asimonov By the looks of the log file it seems you have a pretty old version of the sim try https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.1.0-motion-planning

Also what beta project link are you referring to?

asimonov, 2018-03-12:
yes, that was on the original simulator given for beta-testing of the project. still have to try the final classroom simulator.

asimonov, 2018-03-12:
on the new simulator I get a flickering screen on Mac (10.11.6) and nothing works in the simulator for 3D project.

ryan-keenan, 2018-03-12:
Apparently shift-s for colliders file generation was turned off in the simulator that Dom linked to above.  He just pushed another release last night.  Could you please try this one: https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.1.1-sim
I just double checked on mac 10.13.3 and popup / save colliders.csv works for me.  As for how your planning code works/doesn't work it likely has to do with sim / colliders file changes.  I can help with that if the problem persists

asimonov, 2018-03-12:
I have just downloaded v0.1.1-sim.  and in fact all the latest releases. I can run backyard flyer and controls sections inside simulator. but every time (and in all simulators versions) when I chose Motion Planning the simulator goes into 'wild flickering mode' on my mac (10.11.6). does no one else experience this on Mac?

ryan-keenan, 2018-03-12:
We have not heard this from anyone else that I know of. Do you want to try upgrading your OS and see if it helps?

asimonov, 2018-03-12:
this is strange. i would expect more people to still remain on El Capitain. (10.11.6 is the latest version of El Capitain). i do not want to upgrade my entire OS on my main and only laptop just to test if it solves FCND simulator issue. too much risk of hitting problems with something else.

if there is no way to debug this I may do this one project on Linux. 

but i wonder why this one project within the same app would fail for me only.

domluna, 2018-03-14:
I'm on 10.13.3 (17D102) and I don't have this issue. @ryan-keenan the project available in the classroom via vm right?

ryan-keenan, 2018-03-15:
I'm going to close this for now since it doesn't seem to be widely reported and we provide the colliders file with the project and in classroom.  If you find resolution or that this problem is more widespread feel free to comment on  / reopen this ticket. Thanks!

closed 2018-03-15 by ryan-keenan

asimonov, 2018-03-25:
I have managed to upgrade to High Sierra and can confirm that simulator now
works for 3D planning on my Macbook.

ryan-keenan, 2018-03-26:
Thanks for following up on this!  Seems to be further proof that Apple forces you to upgrade by sabotaging previous versions of things ;)