udacity/fcnd-issue-reports #106: Environment creation error

suraj1988, 2018-02-28:


suraj1988, 2018-02-28:
Hey there,Im on a Windows os. Ive installed git and miniconda., and cloned the starterkit directory. 
An error pops up when im creating the environment (Attached screenshot). It appears a required file is missing.
Any help would be appreciated.Thanks.
[image]

jtuckerdunn, 2018-02-28:
Multiple students have experienced this setup error.

suraj1988, 2018-03-01:
Fixed it. The issue seems to be with the shapely file.  For some reason it only seems to work if i manually download a shapely wheel and then modify the environment. yml file to point to the downloaded wheel. After that i have to update the fcnd from the terminal. Hope that helps if you're trying to find a workaround

jtuckerdunn, 2018-03-02:
Changed shapely pip install to conda install in environment.yml file after reproducing the error on a windows computer and the issue seems to be fixed. Will leave this issue open for more testing.

jtuckerdunn, 2018-04-02:
Fixed

closed 2018-04-02 by jtuckerdunn