udacity/fcnd-issue-reports #107: Backyard Flyer Solution

berndrietberg, 2018-02-28:
Hey Guyz, 
In the solution of Backyard Flyer, during going from waypoint to waypoint (waypoint phase), it denotes on line 48:

if np.linalg.norm(self.target_position[0:2] - self.local_position[0:2]) < 1.0:
print('local position 0,1,2:',self.local_position[0:2])
#go to next waypoint 

meaning it is checking if the drone is in the required target position, and if it is, go to the next waypoint.

However, it is only checking the first two coördinates: self.target_position[0:2] contains coördinate 0 and coördinate 1, but stops before getting to 2.
Thus, it is only checking the x and y, and not the z coördinate. One could have the drone lowering by say 10 meters, and still pass the if statement.
 
Should this be changed to 0:3 ?

jtuckerdunn, 2018-04-23:
BYF solution did not work with the recommended change. There is an issue with the target_position[2] and local_position[2] net matching. Will leave as is.

closed 2018-04-23 by jtuckerdunn

berndrietberg, 2018-06-07:
Correct. The z coordinate is negative, so needs to be calculated seperately.

This is what I did:            

#Calculate if the drone is on target:
            self.distance_to_target = abs(self.local_position[0:3] - self.target_position[0:3])
            #The z coordinate is negative in local_position and positive in target_position, so needs to be calculated seperately:
            self.distance_to_target[2] = abs(-1*self.local_position[2] - self.target_position[2])
#Create a boolean is_on_target:
            self.is_on_target = (self.distance_to_target < 0.5).all()