udacity/fcnd-issue-reports #108: Udacidrone Docs: ValueError: cannot reshape array of size 1 into shape (2) when running the demo Plotting Realtime Data

binliu-base, 2018-02-28:
I ran into a "ValueError: cannot reshape array of size 1 into shape (2)" when try to 
plot the NED position with visdom. 

the visdom-tutorial link
https://udacity.github.io/udacidrone/docs/visdom-tutorial.html

Figure out cause is there are two statements wrong

line: 11:  ne = np.array(self.local_position[0], self.local_position[1]).reshape(-1, 2)
line: 19:  d = np.array(self.local_position[2])

Which should be:
ne = np.array((self.local_position[0], self.local_position[1])).reshape(-1, 2)

and 

d = np.array([self.local_position[2]])

JechoVarta, 2018-02-28:
np.array([self.local_position[0], self.local_position[1]]) 
can be written as 
self.local_position[:2]

JechoVarta, 2018-02-28:
previous comment will not work, must change self.t to a list in the np.array([]) @ line 21

domluna, 2018-03-09:
This should be fixed, reopen if it’s not the case

closed 2018-03-09 by domluna

amay1212, 2018-12-27:
Hi All, 

I'm getting the error like:--

states_next = states_next.reshape((batch_size, ) + self.input_shape)
ValueError: cannot reshape array of size 1001 into shape (1,10,10,10)

Whereas,my batch_size is 1 and input_shape is 1000.

Please help me to fix this.

Thanks!!