udacity/fcnd-issue-reports #117: 3D motion planning - cannot see top-down view of world in sim?

Shreeyak, 2018-02-28:
For this project, it's handy to visualize movement of quad through the entire map. Please provide an interface to zoom out further, such that the entire world is visible, and a method to see the top-down view of the world, as the quad navigates through the city.

I don't even know how far the quad has yet to travel!

Also, the view should not snap back to default when the RMB is released.

ryan-keenan, 2018-03-01:
Hey Shreeyak, I agree an was actually talking to Noam (our Unity guy) about modifying this before I looked at this ticket!  Will be fixed soon and we'll push you guys a new sim to play with.

ryan-keenan, 2018-03-02:
This is implemented now, please check out the sim in the workspaces VM that I linked to in the beta testers channel

closed 2018-03-02 by ryan-keenan