udacity/fcnd-issue-reports #128: 2.1.7 Partial Plan quiz

delvingdeep, 2018-03-03:
I find the options are bit confusing. As per correct answer, we can only expand 2 states out of 3 possibility; but the 3 third possibility which is towards right and ends in obstacle can also be expanded up and down giving use 4 additional partial plans making it 8. 

What I think we should do is to add some clarification to support the current answer or should include new correct answer as one of the options.

ryan-keenan, 2018-03-07:
Thanks for reporting, yes we need to make this more clear.  I'll work on this later this week.

jtuckerdunn, 2018-03-09:
There are many students on slack who are confused by this quiz. I am also adding a comment by @jwdunn1 in issue #87 when this was brought up previously: The updated text is awkward: "But next can only expanding 2 out of those 3 plans because of the obstacle." Perhaps: "That's correct! From the start state, you immediately get 3 partial plans moving up, down, and right. But next, you can only expand 2 out of those 3 plans because of the obstacle. Moving to the right past the obstacle, you'll get one more plan by expanding into the cell between the obstacle and the goal, for a total of 4."

ryan-keenan, 2018-03-10:
Finally got around to fixing this.  There was definitely some misleading language in there about plans occupying the same cells etc.  I've added text and images to explain the answer to the quiz in detail and I think it should be much more clear now! https://classroom.udacity.com/nanodegrees/nd787/parts/5aa0a956-4418-4a41-846f-cb7ea63349b3/modules/0c12632a-b59a-41c1-9694-2b3508f47ce7/lessons/c9b7c43f-6c90-401a-a2ce-ac10613e34f1/concepts/d09207e7-f8d6-40a4-9147-9634d672e7cb?contentVersion=1.0.0&contentLocale=en-us

closed 2018-03-10 by ryan-keenan