udacity/fcnd-issue-reports #135: networkx require

jaketlussier, 2018-03-05:
I get this when I run the solution notebook for Lesson 4 "Probabilistic Roadmap Exercise":

Installing collected packages: decorator, networkx

Successfully installed decorator-4.2.1 networkx-2.1
Cleaning up...
---------------------------------------------------------------------------
VersionConflict                           Traceback (most recent call last)
<ipython-input-4-6aa7f6a21c1a> in <module>()
      2 get_ipython().system('{sys.executable} -m pip install -Iv networkx==2.1')
      3 import pkg_resources
----> 4 pkg_resources.require("networkx==2.1")
      5 import networkx as nx

/opt/conda/lib/python3.6/site-packages/pkg_resources/__init__.py in require(self, *requirements)
    982         included, even if they were already activated in this working set.
    983         """
--> 984         needed = self.resolve(parse_requirements(requirements))
    985 
    986         for dist in needed:

/opt/conda/lib/python3.6/site-packages/pkg_resources/__init__.py in resolve(self, requirements, env, installer, replace_conflicting, extras)
    873                 # Oops, the "best" so far conflicts with a dependency
    874                 dependent_req = required_by[req]
--> 875                 raise VersionConflict(dist, req).with_context(dependent_req)
    876 
    877             # push the new requirements onto the stack

VersionConflict: (networkx 1.11 (/opt/conda/lib/python3.6/site-packages), Requirement.parse('networkx==2.1'))

ryan-keenan, 2018-03-07:
This fails sometimes, need to reset kernel and try again.  Suboptimal but we can't replace this package in the backend yet because it would break stuff in iSDC.  Updates to iSDC coming soon and then we can add this package directly to the backend and import directly.

jtuckerdunn, 2018-04-16:
@ryan-keenan can we add language in the notebook that says to reset the kernel if this error is experienced?

jtuckerdunn, 2018-05-04:
Recent lesson feedback regarding this issue: 

Suggestion for further improvement:\

When I run the jupyter notebook on the Probabilistic Roadmap Exercise page I got the following error: "VersionConflict: (networkx 1.11 (/opt/conda/lib/python3.6/site-packages)\, Requirement.parse('networkx==2.1'))" .\

The jupyter notebook on the solution page works fine.

domluna, 2018-07-13:
@jtuckerdunn Even with that error it looks like it installs fine and the version is correct. I commented out the `pkg_resources` lines so the error doesn't pop up. It's unclear why it doesn't also pop up in the solution notebook.

closed 2018-07-13 by domluna