udacity/fcnd-issue-reports #140: Project 2 motion planning simulator reset

gantners, 2018-03-06:
When testing the backflyer solution in the beginning it is Needed to do multiple resets on the sim with SHIFT+R.
Also the Position various from reset to reset. It Switches between two positions, one in the City one outside

domluna, 2018-03-07:
@demi180 is working on a fix

jwdunn1, 2018-03-09:
Also, after a reset (on the second try) in Motion Planning all the trees are missing. The buildings return, but not the vegetation. (v0.1.0 Simulator and v0.1.0 Motion Planning)

domluna, 2018-03-09:
@jwdunn1 The most recent versions don't have the fix for this yet. I'll post an update here when we have a version that does. Should be soon

jtuckerdunn, 2018-04-16:
@domluna has this issue been fixed?

demi180, 2018-04-17:
Yes, we’ll have a new release by tomorrow.

closed 2018-04-20 by jtuckerdunn

jwdunn1, 2018-05-06:
Has the new release been delayed?