udacity/fcnd-issue-reports #142: 5.11 Quaternions 2:02 does not contain the formula

lxfind, 2018-03-07:
In 5.11 Quaternions 2:02 when it talks about Euler to Quaternion conversions, the slide simply has "WHERE q = [a b c d]" and does not provide the relationship between the three params and the value of a, b, c, d. Not sure what you are trying to get out in that slide. Is the actual formula missing?

ryan-keenan, 2018-03-07:
Thanks for reporting! I'll look into this

ryan-keenan, 2018-03-08:
The idea here is that any quaternion q = a+bi+cj+dk and so the three quaternions representing roll, pitch and yaw are given in the slide as 4-vectors in terms of phi, theta and psi. In each case the elements of those vectors correspond to the a, b, c and d values of the quaternions representing roll, pitch and yaw. Does that make sense?  I can see how this slide could be confusing, but I'm not sure what would make it more clear.

lxfind, 2018-03-08:
How about just write down the formula of the value of a, b, c, d in terms of roll, pitch and yaw in that slide?
i.e. "Euler Angles to Quaternion Conversion" in https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles

ryan-keenan, 2018-03-08:
Well the goal of this slide is not necessarily to write down the composition of rotations about roll/pitch/yaw but rather the individual quaternions that correspond to a roll or a pitch or a yaw maneuver. The intention was to keep things separate for simplicity but maybe there's still a need to add a better explanation.

ryan-keenan, 2018-03-25:
Closing this issue for now.  Please feel free to reopen with new suggestions for improvement!

closed 2018-03-25 by ryan-keenan