udacity/fcnd-issue-reports #143: 4.14 A* Wrong Triangle Inequality Theorem

lxfind, 2018-03-07:
The theorem should use <= instead of <, otherwise Manhattan distance will never be a valid heuristics.
Also refer to the wikipedia pages:
https://en.wikipedia.org/wiki/Triangle_inequality
https://en.wikipedia.org/wiki/Admissible_heuristic

ryan-keenan, 2018-03-07:
Thanks for reporting! I'll look into this

ryan-keenan, 2018-03-08:
Ok I've put in a request to edit that graphic but unfortunately Nick says "less than" instead of "less than or equal" so we'll have to live with it in the audio for now.

ryan-keenan, 2018-03-16:
The video has now  been updated to show "<=" which is all we can do for now. Hopefully isn't too misleading that Nick says "less than" as that appears on screen.

closed 2018-03-16 by ryan-keenan