udacity/fcnd-issue-reports #149: Motion Planning: some colliders missing

jwdunn1, 2018-03-10:
The colliders.csv file is missing the northern-most 100 meters of the city (buildings north of Washington Street) This is easily verified with a scatter plot of the x-y values (in Excel or other plotter tool). The collider yields a grid of 800x900 (rowsXcolumns). Expected 900x900.

ryan-keenan, 2018-03-10:
Thanks for reporting this!  We'll investigate and update on this ticket

ryan-keenan, 2018-03-12:
This is now fixed in the latest release of the simulator (trees now included as colliders too!). `colliders.csv` file has been updated for classroom exercises and the new sim is in the workspace VM as well.  https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.1.1-sim

closed 2018-03-12 by ryan-keenan