udacity/fcnd-issue-reports #151: Lesson 6 -  15 Graph search Excersize issues

karthi1975, 2018-03-10:
Collecting networkx==2.1
  1 location(s) to search for versions of networkx:
  * https://pypi.python.org/simple/networkx/
  Getting page https://pypi.python.org/simple/networkx/
  Looking up "https://pypi.python.org/simple/networkx/" in the cache
  Current age based on date: 197
  Freshness lifetime from max-age: 600
  Freshness lifetime from request max-age: 600
  The response is "fresh", returning cached response
  600 > 197
  Analyzing links from page https://pypi.python.org/simple/networkx/
    Skipping link https://pypi.python.org/packages/01/01/7b7056bc70c410e17a530d5219f64cb2d8ca74569213c4ac90470c819566/networkx-1.7rc1-py2.6.egg#md5=9cf8e0e59ca11297c37dadd1dbd4484c (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/03/66/6f7344240222219b93472b479dfb4f84205a05db9bfb5270fcc7ee5f9c56/networkx-1.4rc1-py3.2.egg#md5=932fa909f25484068f25842ea1e5bbc7 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/0c/d0/120c266d9e6ab1276db7332b92c051e5cc456bf98ed5e967a726135daaf9/networkx-1.11rc2-py2.py3-none-any.whl#md5=ae10a8297e4848115039246c02bb1d82 (from https://pypi.python.org/simple/networkx/), version: 1.11rc2
    Skipping link https://pypi.python.org/packages/0d/6f/9e685de61145e554036b727de3c4074c6d4c8f12bb297929e3de875f1520/networkx-0.36.win32.exe#md5=96d027f31860bf5a6b5a624c0f5616fa (from https://pypi.python.org/simple/networkx/); unsupported archive format: .exe
    Skipping link https://pypi.python.org/packages/0f/de/e75942a51224fea54cd5439c5ffc857cc96e38376fa7be5ba02fdadd146c/networkx-1.8rc1-py3.2.egg#md5=e4d7bdff9286ee4b232dec490d8fdafd (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/11/09/a5bb974e846b03f9b94cc6f8322aec2ed64acd4c76c81ca7226b1ae49e93/networkx-0.99-py2.4.egg#md5=659132ddf83013a72acea973e2b93657 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/11/22/92049588d5cb27a5ef4c8cdc344c0c32421205898fb8c8630152935b2f83/networkx-1.7-py2.7.egg#md5=1d4c59f1e894f39f8928be8718905969 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/11/42/f951cc6838a4dff6ce57211c4d7f8444809ccbe2134179950301e5c4c83c/networkx-2.1.zip#md5=4a2c2a62dfc46ba7d594daca7c834995 (from https://pypi.python.org/simple/networkx/), version: 2.1
    Skipping link https://pypi.python.org/packages/12/8c/f721d617ebced122efed2898369d5e0484e4e042a40c629b6462d132b6f8/networkx-1.3-py2.7.egg#md5=93aaae77abd632a4aa58cf4493f04bf4 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/16/36/77441e90c4baddbed936e609cb731ff15cd9af489d0a373363bebcabcc66/networkx-1.0-py2.6.egg#md5=b543700e4e2959de2b7224c9338def67 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/17/98/363a7269fd60560550f3b8c4906f0b2f07a1813a81ead612f5556a9cd13b/networkx-0.35.1.tar.gz#md5=4f9924488a05bd61ace7dd85a5272732 (from https://pypi.python.org/simple/networkx/), version: 0.35.1
    Skipping link https://pypi.python.org/packages/17/b4/15c0256951098bc7e5b25111a5f3c99d2b6fdcf0d4026c62e1559331fd2d/networkx-1.11-py3.4.egg#md5=b6bc6a463176a4165deddc60b332612c (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/1b/7b/616f4c979b3843a8a2a9ee77a097e34333f78b11d6a97bebc963ddfd38e2/networkx-0.34.tar.gz#md5=24032de84e97407bbabf030e6c70ce75 (from https://pypi.python.org/simple/networkx/), version: 0.34
    Found link https://pypi.python.org/packages/1e/ce/c5f0d432ab38b114a18020c1f716e08347b580b6865111ef1314995627a2/networkx-1.8.1.tar.gz#md5=b4a9e68ecd1b0164446ee432d2e20bd0 (from https://pypi.python.org/simple/networkx/), version: 1.8.1
    Skipping link https://pypi.python.org/packages/20/4f/28a6b34c8d6160581a96e496f0311fbc0f77714823c1c139243f5b2c4c25/networkx-1.3rc1-py2.7.egg#md5=b1b7a45ab699c5c5f332046ff8f3402d (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/23/cc/3081f2d6d65dd47d967212d29c4e8f483308b027b704d3dc5b59a495e87c/networkx-1.10-py3.4.egg#md5=5cb1a64e7512efb32233e5683c9fc741 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/23/e0/3c401ef8475477a2bd9896f1524d2b22888c892e414201c0ddc7e6cbfec9/networkx-1.5-py2.6.egg#md5=7c8f544b7d7218d91836bbb900f0b7f0 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/24/63/38d4560635fde1a8336c632de734aa18783a2668e01bcac64b281442ebba/networkx-1.8rc1.tar.gz#md5=fdde72adcbc11366a6bcc338c1d5aecd (from https://pypi.python.org/simple/networkx/), version: 1.8rc1
    Skipping link https://pypi.python.org/packages/24/7c/06aa502af424ff6896b408d6bd63d481840262818d26161819856cbcc020/networkx-1.3-py2.6.egg#md5=504cfdf0e3d1511c818b89e2502bdb93 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/27/92/bb9f75dda0ab7b64ba97089f8f494c22117901a468cef483ea6879ca8299/networkx-1.5rc1-py3.1.egg#md5=898ce95c1f635e6d843a67ddc8aa967e (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/29/12/8aca3f5d87e55720f0e4e19f8c3c860431d2b44241dede3a350561249655/networkx-1.5rc1.zip#md5=43d85f49888bfa682869ef46851f5221 (from https://pypi.python.org/simple/networkx/), version: 1.5rc1
    Skipping link https://pypi.python.org/packages/2a/59/ec4c735f4216eb12380393d6cb972611b391556345c693c1a91bf2945335/networkx-1.11rc2-py2.7.egg#md5=2983c32b63ae8646fe3ffdd1e3ec9f15 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/2a/f1/2a0290386ec43ec6874ddb469b653cf02cba1ce13c519f09464a25f43917/networkx-1.8-py2.6.egg#md5=1759ce2e3652c26de026b13f5d5c2c17 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/30/ff/35f2fd9d90cfe7e6fdb1f8da212c21df325bcbee90e980a567b09c7c6ee7/networkx-0.34-py2.5.egg#md5=c540d9c6203b303bb8d9cf97e1d527e1 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/32/a1/ac289f24908c7f721ae37078e60bae725d0e45428acd83dadec52a035abc/networkx-1.9.1.zip#md5=9c70edab46a9ce64ed03e4bbf15f2c2b (from https://pypi.python.org/simple/networkx/), version: 1.9.1
    Skipping link https://pypi.python.org/packages/32/fa/9b43a508a89b8c3476ab5faf071d1169983f2d496b39dbccfc719547131b/networkx-1.8.1-py3.3.egg#md5=f6842885f9ef2afa20f667d62bf94d17 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/36/fe/776a47a05d349049646d9e264fdffc98834706ea33eabecf7beac8a7af5c/networkx-1.0rc1-py2.5.egg#md5=5148e1cde10c341b47fcfcf1767adfec (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/37/a8/4a863e0240c98b344b1b26d4c57bc0b1c392aba8b7e6eaa477849d80b1f6/networkx-0.35.win32.exe#md5=0222cf3dbf1346c6950471e724ddd3a3 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .exe
    Skipping link https://pypi.python.org/packages/37/c1/53456e44e20119266985dec0fa05542e79388f9aa129329091b61c342104/networkx-0.35.1-py2.4.egg#md5=a6382079e85eda3179879cea77751def (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Skipping link https://pypi.python.org/packages/3a/04/5888ecf7b1a7b93641b9251cc63f3155f382c4b7193f7c6fd7c26679d4b0/networkx-1.3rc1-py3.1.egg#md5=505e3b1754b4aaffbc62a13997317d48 (from https://pypi.python.org/simple/networkx/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/3b/d4/1b6ca16678d13ffd426bcd199231f3d5effaa6d234747fb88899e649dec3/networkx-1.11.zip#md5=c7690890aa28eb2e061f9e50c20ebada (from https://pypi.python.org/simple/networkx/), version: 1.11
[... cut ...]
/opt/conda/lib/python3.6/site-packages/pkg_resources/__init__.py in resolve(self, requirements, env, installer, replace_conflicting, extras)
    873                 # Oops, the "best" so far conflicts with a dependency
    874                 dependent_req = required_by[req]
--> 875                 raise VersionConflict(dist, req).with_context(dependent_req)
    876 
    877             # push the new requirements onto the stack

VersionConflict: (networkx 1.11 (/opt/conda/lib/python3.6/site-packages), Requirement.parse('networkx==2.1'))

ryan-keenan, 2018-03-10:
Thanks for reporting this!  Unfortunately this import seems not to work on occasion, especially if you've already done some work in the notebook and are rerunning from the top. We will be upgrading to networkx 2.1 in the backend soon, but for now we need the import because it breaks other classroom quizzes to switch away from 1.11.  Please select "restart and clear output" from the "Kernel" menu and give it another try.  Sorry for the clunkiness!

ryan-keenan, 2018-03-10:
I'll add a note to the exercises that use this import that you might need to run it a couple times

closed 2018-03-10 by ryan-keenan

mehmetyldz87, 2018-03-17:
Dear Ryan

when I executed those codes. I also tried to add points=[] into grid.py but I did not fix it
 
# This is now the routine using Voronoi
grid, edges = create_grid_and_edges(data, drone_altitude, safety_distance)
print(len(edges))

-----------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-8-5a818c19d39d> in <module>()
      1 # This is now the routine using Voronoi
----> 2 grid, edges = create_grid_and_edges(data, drone_altitude, safety_distance)
      3 print(len(edges))

/home/workspace/grid.py in create_grid_and_edges(data, drone_altitude, safety_distance)
     77 
     78             # add center of obstacles to points list
---> 79             points.append([north - north_min, east - east_min])
     80 
     81     # TODO: create a voronoi graph based on

NameError: name 'points' is not defined

mehmetyldz87, 2018-03-19:
Fixed thanks