udacity/fcnd-issue-reports #154: Bresenham exercise solution does involve floating point computation

lnson, 2018-03-11:
In the second snippet, m is a float; therefore, the implementation does involve floating point computation, which according to Lesson 6.7 Bresenham algorithm eliminates.
 
def bres(p1, p2): 
    """
    Note this solution requires `x1` < `x2` and `y1` < `y2`.
    """
    x1, y1 = p1
    x2, y2 = p2
    cells = []
    
    m = (y2 - y1) / (x2 - x1)
    print(m)  # m is a float
    
    line_val = y1
    i = x1
    j = y1
    
    while i < x2:
        cells.append([i, j])
        if line_val + m > j + 1:
            j += 1
        else:
            line_val += m
            i += 1
        
    return np.array(cells)
cells = bres((0,0), (8, 21))

ryan-keenan, 2018-03-15:
Good catch! In fact Bresenham's algorithm is not completely free of floating point operations because the slope is a floating point value.  However, it is still much more efficient than other ray tracing algorithms because apart from the slope (which only needs to be calculated once), everything else is integer.

closed 2018-03-15 by ryan-keenan

lnson, 2018-03-15:
I think it can be made FP free. Here is how i implemented it:
```python
def bres(p1, p2):
    """
    Note this solution requires `x1` < `x2` and `y1` < `y2`.
    """
    x1, y1 = p1
    x2, y2 = p2
    cells = []
    
    dx, dy = x2 - x1, y2 - y1

    # (x + 1) dy / dx < y + 1 ==> x dy - y dx < dx - dy
    # Let d = x dy - y dx.
    d = 0
    while x1 < x2 and y1 < y2:
        cells.append([x1, y1])
        if d < dx - dy:
            d += dy
            x1 += 1
        else:
            d -= dx
            y1 += 1

    return np.array(cells)
```

ryan-keenan, 2018-03-15:
Very cool! I guess I didn't dig deep enough on how to make this solution FP free. I've replaced the solution in the classroom with your solution slightly modified to better handle the case where dx == dy (or other instances where the line passes directly through a corner).  Here it is, let me know what you think 
```
def bres(p1, p2):
    """
    Note this solution requires `x1` < `x2` and `y1` < `y2`.
    """
    x1, y1 = p1
    x2, y2 = p2
    cells = []
    
    dx, dy = x2 - x1, y2 - y1

    # (x + 1) dy / dx < y + 1 ==> x dy - y dx < dx - dy
    # Let d = x dy - y dx.
    d = 0
    i = x1
    j = y1
    
    while i < x2 and j < y2:
        cells.append([i, j])
        if d < dx - dy:
            d += dy
            i += 1
        elif d == dx - dy:
            # uncomment these two lines for conservative approach
            #cells.append([i+1, j])
            #cells.append([i, j+1])
            d += dy
            i += 1  
            d -= dx
            j += 1
        else:
            d -= dx
            j += 1

    return np.array(cells)
```

lnson, 2018-03-15:
Ah yes I missed those edge cases. Thank you for fixing them. This looks great to me.