udacity/fcnd-issue-reports #159: Controls Project - Increase run time of 01_intro scenario

jtuckerdunn, 2018-03-13:
The current runtime is 1 second. Should be increased to around 3 seconds since we are telling students to adjust the mass parameter and see if it is roughly hovering.

adrnp, 2018-03-13:
I think 1 sec is enough to see whether or not it's staying roughly in place.  If you play around with the mass parameter, if you're off by even 0.1 it'll either drop pretty quickly or rise pretty quickly.  Also, since there is no control on the drone (besides equal thrust to all motors to match the mass), in 3 seconds it'll do some pretty crazy stuff (won't actually hover since there is no control in the loop trying to keep level, etc).

jtuckerdunn, 2018-03-19:
ok

closed 2018-03-19 by jtuckerdunn