udacity/fcnd-issue-reports #161: motion-planning-issue

karthi1975, 2018-03-14:
Today , I have taken latest version of code  and simulator for mac. I am getting following issue. (Previous version and simulator worked just fine).
 When I try running  below  and getting following error

python motion_planning.py 
Logs/TLog.txt
Logs/NavLog.txt
starting connection
arming transition
Searching for a path ...
global home [-122.39745   37.79248    0.     ], position [-1.22397452e+02  3.77924804e+01  2.00000000e-02], local position [ 0.05165228 -0.16657087 -0.02034687]
Traceback (most recent call last):
  File "/Users/karthi/anaconda3/envs/fcnd/lib/python3.6/site-packages/udacidrone/drone.py", line 277, in notify_callbacks
    fn()
  File "motion_planning.py", line 67, in state_callback
    self.plan_path()
  File "motion_planning.py", line 133, in plan_path
    data = np.loadtxt('colliders.csv', delimiter=',', dtype='Float64', skiprows=3)
  File "/Users/karthi/anaconda3/envs/fcnd/lib/python3.6/site-packages/numpy/lib/npyio.py", line 1037, in loadtxt
    next(fh)
StopIteration
takeoff transition
waypoint transition
Traceback (most recent call last):
  File "/Users/karthi/anaconda3/envs/fcnd/lib/python3.6/site-packages/udacidrone/drone.py", line 277, in notify_callbacks
    fn()
  File "motion_planning.py", line 46, in local_position_callback
    self.waypoint_transition()
  File "motion_planning.py", line 88, in waypoint_transition
    self.target_position = self.waypoints.pop(0)
IndexError: pop from empty list
landing transition
disarm transition
manual transition
Closing connection ...

karthi1975, 2018-03-14:
The colliders.csv is the same folder strucutre of motion_planning.py.

ryan-keenan, 2018-03-14:
It looks like you're not finding a path so `self.waypoints.pop(0)`. I would recommend breaking out each of these steps into a separate jupyter notebook so you can diagnose the problem.  There were some minor changes to the code in the recent updates that may be causing your path planner to fail.

karthi1975, 2018-03-14:
I am just trying to run the off-self code motion-planning.py to test . IT fails. May be I am not properly loading colliders. ? I did shift-s to load colliders ..In the simulator  I pointed the directory the colliders and did save them ...and still not working. I have uploaded the  screen shot.  I would like see (original )motion_planning.py to working and then I can add code incrementally.
[image]

ryan-keenan, 2018-03-14:
Ah ok I think I found the problem, sign flip error.  Please pull the new changes and try again. 
Sorry about that!

closed 2018-03-14 by ryan-keenan

karthi1975, 2018-03-14:
Great! Thanks it works now.