udacity/fcnd-issue-reports #165: skiprows not updated when colliders.csv changed format

Mark70117, 2018-03-15:


Mark70117, 2018-03-15:
data = np.loadtxt('colliders.csv', delimiter=',', dtype='Float64', skiprows=3)
https://github.com/udacity/FCND-Motion-Planning/blob/6b14950b9ddcae940126b1eb87db6d4716e68306/motion_planning.py#L134

colliders was changed, but not this line which skips past the "posX" header
so now first colliders is missed

skips first row of data.  colliders.csv  used to start
lat0,lon0
37.792480,-122.397450
posX,posY,posZ,halfSizeX,halfSizeY,halfSizeZ
-305,-435,85.5,5,5,85.5

now starts
lat0 37.792480, lon0 -122.397450
posX,posY,posZ,halfSizeX,halfSizeY,halfSizeZ
-310.2389,-439.2315,85.5,5,5,85.5

ryan-keenan, 2018-03-15:
Great catch! Just pushed this fix now thanks.

closed 2018-03-15 by ryan-keenan