udacity/fcnd-issue-reports #168: Error in Lesson 8.7 Dubins Car Exercise

tjphoton, 2018-03-17:
Lesson 8.7 Dubins Car Exercise, in the Dubins-Car-Solution.ipynb, 
Should `ntheta = theta + v*np.tan(angle)*dt` be `ntheta = theta + np.tan(angle)*dt`?
The function name "simulate" in cell 6 is not consistent with cell 7 "dubins_car_model"

domluna, 2018-03-18:
The function name has been updated to `simulate`.

If v=1 then they're equivalent. In general, though there are many versions of Dubin's car model, some even take into account the length and width of the vehicle.

closed 2018-03-18 by domluna

jwdunn1, 2018-04-03:
The source of confusion is the top cell expression "\dot{\theta} = tan(u)" should read: "\dot{\theta} = v * tan(u)" (assuming L = 1.0 [the wheelbase]) The equation is missing the velocity component, yet it is implemented with "v" in the solution for theta. Equation reference: http://www.cs.cmu.edu/~motionplanning/reading/PlanningforDynamicVeh-1.pdf

ryan-keenan, 2018-04-07:
Thanks for reporting this guys! I finally went in and changed the notebook to say `\dot{\theta} = v * tan(u)`