udacity/fcnd-issue-reports #169: Lesson 8.9 Steering Exercise Solution Errors

tjphoton, 2018-03-17:
Lesson 8.9 Steering Exercise, in Steering-Solution.ipynb,

1. In simulate(), should 
`ntheta = theta + v*np.tan(angle)*dt` be 
`ntheta = theta + np.tan(angle)*dt`?

2. In steer(), `np.arctan2(x2[1] - x1[1], x2[0] - x1[0]) - theta` is the heading correction, not the control input. 
Based on Dubin's car dynamic model, steering angle should be
```phi = np.arctan2(x2[1] - x1[1], x2[0] - x1[0]) - theta
angle = np.arctan(phi)
angle = np.clip(angle, -MAX_STEERING_ANGLE, MAX_STEERING_ANGLE)```
This shouldn't have a huge effect on the result though, since $\theta$ and tan($\theta$) are approximately equal for $\theta$ < 30 degree

3. `v = np.linalg.norm(np.array(x1[:2]) - np.array(x2[:2]))` is the distance, not velocity. There should be a time variable in the denominator, which depends on the remaining time `total_time`, and discrete time step `dt`.

tjphoton, 2018-03-17:
Correction:
```
angle = np.arctan(phi/dt)
```

tjphoton, 2018-03-17:
take back "This shouldn't have a huge effect on the result" part, because dt is not necessary equals to 1.

domluna, 2018-03-18:
The notebook is being updated to be more akin to what's presented in the video, particularly the connection with the batwing figure. It should be much clearer.

ryan-keenan, 2018-04-07:
I think this issue is closed, but please reopen if problems persist!

closed 2018-04-07 by ryan-keenan