udacity/fcnd-issue-reports #170: Lesson 8.11 RRT Exercise Errors

tjphoton, 2018-03-17:
Lesson 8.11 RRT Exercise, in RRT-Solution.ipynb,
1. sklearn package is not included in the fcnd starter environment. 
need to do the `# !{sys.executable} -m pip install sklearn` trick to have it installed.

2. In the obstacle visualization code cell `plt.imshow(grid, cmap='Greys', origin='upper')`, `upper` should be changed to `lower` to be consistent with the final visualization.

3. In the code cell 11, it only check whether the new state is located on an obstacle, but did not check whether the new state is out of grid boundary, nor check whether the new edge to be added will pass through obstacles.

4. Again, obstacle boundary index issue.  When marked obstacles `grid[30:40, 60:82] = 1`,  there are 10 grid cell index marked in the x direction: [30, 31, 32, 33, 34, 35, 36, 37, 38, 39].  In the visualization with `plt.imshow(grid)`, region between 29.5 and 29.5 will be colored black. This caused feasible states with x axis between 29.5 and 30 is misleadingly in obstacle region.

tjphoton, 2018-03-17:
[image]
[image]

tjphoton, 2018-03-17:
Correction: 10 grid cell index marked in the *y* direction

domluna, 2018-04-21:
@tjphoton 

1) scikit-learn has been added https://github.com/udacity/FCND-Term1-Starter-Kit/blob/master/environment.yml#L14
2) done 👍 
3/4) I don't think there's much we can do about how matplotlib plots things. We can use `int(np.round(x))` instead of `int(x)`. This would be more in line with how matplotlib plots things. However, since we're not checking if there's an obstacle between 2 points there's still the possibly a line will be drawn through the corner of an obstacle.