udacity/fcnd-issue-reports #173: Error running notebook in 6.15

matthewcheok, 2018-03-18:


matthewcheok, 2018-03-18:
Getting following errors:

```
Collecting networkx==2.1
Collecting decorator>=4.1.0 (from networkx==2.1)
  Using cached decorator-4.2.1-py2.py3-none-any.whl
Installing collected packages: decorator, networkx
Successfully installed decorator-4.2.1 networkx-2.1
You are using pip version 9.0.1, however version 9.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
---------------------------------------------------------------------------
VersionConflict                           Traceback (most recent call last)
<ipython-input-4-4a8500274992> in <module>()
      8 get_ipython().system('{sys.executable} -m pip install -I networkx==2.1')
      9 import pkg_resources
---> 10 pkg_resources.require("networkx==2.1")
     11 import networkx as nx

/opt/conda/lib/python3.6/site-packages/pkg_resources/__init__.py in require(self, *requirements)
    982         included, even if they were already activated in this working set.
    983         """
--> 984         needed = self.resolve(parse_requirements(requirements))
    985 
    986         for dist in needed:

/opt/conda/lib/python3.6/site-packages/pkg_resources/__init__.py in resolve(self, requirements, env, installer, replace_conflicting, extras)
    873                 # Oops, the "best" so far conflicts with a dependency
    874                 dependent_req = required_by[req]
--> 875                 raise VersionConflict(dist, req).with_context(dependent_req)
    876 
    877             # push the new requirements onto the stack

VersionConflict: (networkx 1.11 (/opt/conda/lib/python3.6/site-packages), Requirement.parse('networkx==2.1'))
```

ryan-keenan, 2018-03-19:
Hi Matthew, sorry this install seems to fail sometimes. We'll update the version in the backend soon but right now it would break some other things so we have to install manually like this.  If you go to the "Kernel" menu and "Restart and clear output" and try again it should work (or eventually work.

closed 2018-03-19 by ryan-keenan