udacity/fcnd-issue-reports #176: environment.yml - should have commit hash to udacidrone.git

carlosgalvezp, 2018-03-19:
In the FCND-Term1-Starter-Kit repo.

The dependency of udacidrone should specify a commit hash as well; otherwise everywhere will have a different version of the environment.

It also provides a much cleaner way of communicating when the environment needs to be updated.

domluna, 2018-04-21:
@carlosgalvezp I added a note about upgrading udacidrone to the starter kit repo. https://github.com/udacity/FCND-Term1-Starter-Kit/blob/master/README.md.

carlosgalvezp, 2018-04-22:
@domluna I noticed that udacidrone has some version number, currently 0.3.0. Would this be a more flexible solution than using the commit hash? 

As I said, the main motivation is clarity in the communication, as well as having some control of the environment to make it reproducible.

This is just a suggestion more than a bug, I understand if other bugs are more urgent than this ;)

domluna, 2018-07-13:
I'm skeptical to assign a particular hash while the library is still in development since it would just become quickly outdated anyway. We should revisit this when all the content is settled.

For now, the solution is just to upgrade, not ideal, but for now, it works.

closed 2018-07-13 by domluna