udacity/fcnd-issue-reports #179: Wrong test for checking if point inside a convex polygon

haragg2, 2018-03-21:


haragg2, 2018-03-21:
In 2.5D Maps lecture Nick told to arrange the polygon points in counterclockwise order and find the angle (using dot product) between the given point and two successive points of polygon. He said for all such pair of points on polygon the angle will be positive if it's inside the polygon. But it's not correct.
It should be checked using cross product.

ryan-keenan, 2018-04-08:
Thanks for catching this! I've now cut the portion of the video where Nick goes through the description of calculating angles.  As you point out, using the dot product doesn't work for this.  In the exercise we end up using the Python shapely package so doesn't seem like we need this description anyway.  So fixed by removal!

closed 2018-04-08 by ryan-keenan