udacity/fcnd-issue-reports #187: Incorrect A* implementation in A* Exercise

mkontz, 2018-03-22:
I am in the Flying Car Nanodegree program and I believe there are a series of errors in the A* Exercise.

Using standard A* notation f = g + h, were g is the accumulated cost to the end of branch, h is the heuristic from the end of the branch to the goal and f is the sum of g and h and is used to choose which node to explore next.

In the A* Exercise write up and solution, it says the cost is cnew=c+g()+h() . The problem that this new cost is put in both the priority queue to sort which node to is used next and is put in the branch dictionary as the new cost. The new cost for the new branch should be the last cost plus the move or action cost without the heuristic.

In my solution I get the same answer in terms of path, but I don't feed the heuristic into the accumulated cost. Therefore my cost is 10 because each action has a cost of 1 and there are 10 moves required. The path cost for the solution is 34.5629686047 since the heuristic is added to the accumulated cost (g) at each step.

In other situations I am pretty sure this error would negatively affect both the path and the cost.

mkontz, 2018-03-22:
[L4_Planning_As_Search.zip](https://waffleio-direct-uploads-production.s3.amazonaws.com/uploads/5ab42f981171c10011f47693/125516c66e82c728ace21e0d46c4c0a5138e82a781e0b1449b7ce3ab1a4326612a0ce1673295f127b54778154d555ca9105e5e47f4f88b22a4bf69788b5c0befd63e0c22ce3e13bea4905dee5fee7fd616cfe3fe9846d2ea3d2fe68e8d4e391b0d1d8cf75e32a18b06.zip)

mkontz, 2018-03-22:
Uploaded my solution with correct cost as well as provided solution with path cost print statements added.

ryan-keenan, 2018-03-26:
Thanks for catching this and reporting! I've gone through and fixed the instances of A* that appear in the classroom and project.  Keep the great feedback coming!

closed 2018-03-26 by ryan-keenan