udacity/fcnd-issue-reports #191: Simulator for project 2 failed to load sqlite3.dll in windows 10 system

xcarnd, 2018-03-23:
When I run simulator for project 2, I'll get an empty world except the drone and the ground, all the buidlings, lane, etc. are missing.

Occasionally I checked the Unity player log, I found a line like this:

```Plugins: Failed to load 'C:/fcnd/windows/FCND-Sim_Windows_64-bit_Data/Plugins/sqlite3.dll' with error '动态链接库(DLL)初始化例程失败。
'.```

There're Chinese in it because I'm running Chinese windows 10 system. It means "failed to initialize dynamic link library".

It turns out that failure exactly make all the objects missing in simulator. After downloaded another sqlite3 dll binary from sqlite.org and replace the bundled one, the simulator for project 2 work again.

domluna, 2018-03-23:
@xcarnd Thanks for reporting this, great detective work 👍 @demi180 any idea what might be going on here? My guess is mapbox is calling sqlite or something along those lines.

demi180, 2018-03-23:
@xcarnd are you on 64-bit Windows or 32-bit? And which binary were you running?

xcarnd, 2018-03-25:
@demi180  64bit win10 system.

jtuckerdunn, 2018-04-16:
@demi180 any update on this?

closed 2018-07-13 by domluna