udacity/fcnd-issue-reports #192: Lesson 10.16 Uncontrolled Drone No TODO

tjphoton, 2018-03-24:
Looks like solution is identical to student version. TODO is not masked.

ryan-keenan, 2018-03-25:
Please provide a link or more information in this ticket.  Not sure what you're referencing here.

xcarnd, 2018-03-25:
It looks like the notebook is already finished. Here's the link for the classroom. https://classroom.udacity.com/nanodegrees/nd787/parts/5aa0a956-4418-4a41-846f-cb7ea63349b3/modules/b78ec22c-5afe-444b-8719-b390bd2b2988/lessons/4cd7a5c4-12ad-4853-86cb-759d3833e856/concepts/55be0949-78cc-42c1-9812-a724f01c9c06#

jwdunn1, 2018-03-26:
+1 Is this section just informational? Is it supposed to be a challenge? There is no TODO indicated.

AndyAtUdacity, 2018-03-28:
Whoops! Thanks for catching this. It's been fixed!

closed 2018-03-28 by AndyAtUdacity