udacity/fcnd-issue-reports #206: sklearn missing

asimonov, 2018-03-27:
sklearn used in probabilistic roadmap solution but missing in the envrionment

ryan-keenan, 2018-04-07:
Thanks for pointing this out. I just added it to the `environment.yml` file

closed 2018-04-07 by ryan-keenan

ywiyogo, 2018-04-16:
After updating the `environment.yaml` file and deactivate the conda environment using `source deactivate`,  I cannot install the `sklearn`:
```
Solving environment: failed

ResolvePackageNotFound: 
  - sklearn==0.19.1
```

Afterwards, I did run `conda install sklearn==0.19.1`, and got this message:
```
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - sklearn==0.19.1

Current channels:

  - https://repo.continuum.io/pkgs/main/linux-64
  - https://repo.continuum.io/pkgs/main/noarch
  - https://repo.continuum.io/pkgs/free/linux-64
  - https://repo.continuum.io/pkgs/free/noarch
  - https://repo.continuum.io/pkgs/r/linux-64
  - https://repo.continuum.io/pkgs/r/noarch
  - https://repo.continuum.io/pkgs/pro/linux-64
  - https://repo.continuum.io/pkgs/pro/noarch
```
It seems the 0.19.1 version does not exist in those repositories.

ryan-keenan, 2018-04-16:
Aack! sorry about that! Fixed now ;)