udacity/fcnd-issue-reports #217: A_star diagonal

berndrietberg, 2018-03-31:
If one would like to add diagonal movements in a_star, one has to change the valid_actions function. Here, one wants to do if-statements to check if a diagonal action is allowed.
In the valid_actions build provided by one of the solution models, it states: 
"
if (x - 1 < 0 and y + 1 > m) or grid[x - 1, y + 1] == 1:
valid_actions.remove(Action.NORTH)
"

However, 
this implementation has errors in it. This is because the first part checks if the action is out of bounds. But if it is, the "grid[x-1, y+1]"- part will still also be run by the code. Thus, when we're out of bounds, the first part will return 'true'
and the second part will return ERROR OUT OF BOUNDS -> terminate program.
To solve this, I'll use if - else statements:

    if (x - 1 < 0 or y + 1 > m): #if any axis out of bounds
        valid_actions.remove(Action.NORTH_EAST)
    else: #if not out of bounds, check if next step is actually possible:
        if grid[x - 1, y + 1] == 1: 
            valid_actions.remove(Action.NORTH_EAST)
    if (x + 1 > n or y - 1 < 0):
        valid_actions.remove(Action.SOUTH_WEST)
    else:
        if grid[x + 1, y - 1] == 1:
            valid_actions.remove(Action.SOUTH_WEST)
    if (x + 1 > n or y + 1 > m):
        valid_actions.remove(Action.SOUTH_EAST)
    else:
        if grid[x + 1, y + 1] == 1:
            valid_actions.remove(Action.SOUTH_EAST)
    if (x - 1 < 0 or y - 1 < 0):
        valid_actions.remove(Action.NORTH_WEST)
    else:
        if grid[x - 1, y - 1] == 1:
            valid_actions.remove(Action.NORTH_WEST)

jwdunn1, 2018-04-04:
The Python `or` statement short circuits if the first value is true. Thus, the second part of the `or` is **not** executed and no error is thrown. From the docs: "The expression `x or y` first evaluates _x_; if _x_ is true, its value is returned; otherwise, `y` is evaluated and the resulting value is returned." See: https://docs.python.org/3/reference/expressions.html#or and https://stackoverflow.com/questions/2580136/does-python-support-short-circuiting

berndrietberg, 2018-04-05:
Dear JWDunn,

You are correct.
Still, it runs in my code with an error, and without error in my implementation.
It could also be in the 'and' statement.
For example:
if (x - 1 < 0 and y + 1 > m) or grid[x - 1, y + 1] == 1:
x-1 could be out of bounds, but y+1 could be perfectly within bounds.
Thus (x - 1 < 0 and y + 1 > m) would return False, resulting in executing the second 'grid[x - 1, y + 1]' part. This would then hold an error.
In my implementation, I switched 'and' to 'or': 
'if (x - 1 < 0 or y + 1 > m)'

It makes sense to remove an action if either one is out of bounds.

A simple switch from 'and' to 'or' would probably solve this then.

ryan-keenan, 2018-04-07:
Thanks for your input here! Sounds like this is resolved, but if there's still an issue please feel free to reopen this ticket.

closed 2018-04-07 by ryan-keenan

berndrietberg, 2018-04-09:
Ahhh, so that is where you remove those. 
Yes, a matter of implementation. Nice discussion, thanks!